sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #3: invalid syntax

shuffleman319, 2016-03-28:
tcpSerSock.bind(192.168.0.108)    # Bind the IP address and port number of the server. 
                            ^
SyntaxError: invalid syntax

sunfounder, 2016-05-26:
Sorry for being late. Please post the whole error information, so that we can check where the problem is. And try pull the newest one, and rerun again. Thank you.

notverygoodatthis, 2016-07-13:
invalid syntax when i try to run win_py module.
colon after motor

def run(event):
 global runbtn
 print 'motor', runbtn﻿

javierrasp2016, 2016-09-11:
Hi, please some help with: remote: 
Repository not found.
fatal: repository 'https://github.com/sunfounder/Sunfounder_Smart_video_Car_Kit_for_RaspberryPi.git/' not found

javierrasp2016, 2016-09-11:
I got it. Between remote terminal and terminal. 
Now I have problems withrunning cli_client.py: sudo  python cali_client.py
I have the following error: Socket.error: [Errno 111] Connection refured

sunfounder, 2016-09-12:
Hi, @javierrasp2016 . Have you edited the IP address in cali_client.py? Or have you run the cali_server?

sunfounder, 2016-09-12:
Dear Javier,

Thanks for contacting us. My technical colleague has replied to you on github. Please check on github. Thank you for your understanding.

Best regards,

Jake

support@sunfounder.com
 
From: javierrasp2016
Date: 2016-09-12 02:30
To: sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi
CC: SunFounder; Comment
Subject: Re: [sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi] invalid syntax (#3)
I got it. Between remote terminal and terminal. 
Now I have problems withrunning cli_client.py: sudo python cali_client.py
I have the following error: Socket.error: [Errno 111] Connection refured

javierrasp2016, 2016-09-12:
Hi,
Thanks for answering. I am a starter. I have a raspberry pi connected to a screen. And my laptop working with Putty opening a terminal through the Ip addres from my raspberry pI.

I am following the guide that came with the smart car kit and I am using it on raspberry pi 3. I edited the ip address replacing it by the one in my raspberri (as shown on the guide). After that I tried to run cali_client.py and the error appears.

Also I tried from the raspberry and the error was 111 related to the socket.

Not sure which one should be the client my raspberry pi? Or my laptop?
(Also in a previous step ther option to use client or server and I used the client option. And not sure if I did well).

Many thanks for your help. I like the kit a lot and I would like to get it working.

Best regards

Javier

-------- Original message --------
From: SunFounder
Date:12/09/2016 02:41 (GMT+00:00)
To: sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi
Cc: javierrasp2016 ,Mention
Subject: Re: [sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi] invalid syntax (#3)

Hi, @javierrasp2016<https://github.com/javierrasp2016> . Have you edited the IP address in cali_client.py? Or have you run the cali_server?

-

sunfounder, 2016-09-12:
Sorry about that, we didn't say that clear on the manual. That is not an option, the For Server and For client, you need to do both to make the car work. And, also, the Raspberry Pi is the server side, and your laptop is client side. So you need to run the server on the Raspberry Pi and when your Raspberry Pi is ready and print out(Waiting for connection), you run the client on your laptop, to connect to the server.

Also, I notice that you have a monitor for the Raspberry Pi. and also remotely login to it with PuTTy on your laptop, that make me think that, you may not run the client on your laptop, but run it on the remotely-login-to-your-Raspberry-Pi PuTTy. 
So here's the thing, you can't do that. And if you do, sooner or later you are gonna get into another error named: _tkinter.TclError. The client on the manual, is for Linux PC, not for Windows. For Windows, you may need to download and install Python 2.7 on your laptop, download the code from github, and run cali_client.py with Python IDLE not PuTTy. You can see more details in here under the _tkinter.TclError: https://github.com/sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi#tkinter.tclerror and also see a video about it: https://www.youtube.com/watch?v=QdWtj8qBnug

So, if you still got any problems, you are welcome to ask in here.

javierrasp2016, 2016-09-13:
Hi all,

First of all thak you for your help.

I tried to follow your instruction and trying to follow the video.

Doing For server (in my raspberry and monitor):

pi@raspberrypi:~/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server $ sudo python cali_server.py
Traceback (most recent call last):
  File "cali_server.py", line 5, in <module>
    import motor
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/motor.py", line 24, in <module>
    p = pwm.PWM()
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 55, in __init__
    self.set_all_value(0, 0)
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 124, in set_all_value
    self._write_byte_data(self._ALL_LED_ON_L, on & 0xFF)
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 68, in _write_byte_data
    self.bus.write_byte_data(self.address, reg, value)
IOError: [Errno 5] Input/output error

Doing For client: this is what I´ve got:
Python 2.7.12 (v2.7.12:d33e0cf91556, Jun 27 2016, 15:19:22) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>>
 RESTART: C:\Users\equipo\Downloads\RaspberryPI - Smart CAR\Sunfounder\Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi-master\client\cali_client.py

Traceback (most recent call last):
  File "C:\Users\equipo\Downloads\RaspberryPI - Smart CAR\Sunfounder\Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi-master\client\cali_client.py", line 16, in <module>
    tcpCliSock.connect(ADDR)                    # Connect with the server
  File "C:\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 10061] No se puede establecer una conexión ya que el equipo de destino denegó expresamente dicha conexión
>>>

Please help me to complete the project.

Many thanks.

Javier.

sunfounder, 2016-09-14:
Hi, @javierrasp2016.
It seems like your Pi cannot recognize the servo controller.
Try run this command in terminal:
if your pi is Pi 1 or Pi Zero:
`i2cdetect -y 0`
if your pi is Pi B+, Pi 2 or Pi 3:
`i2cdetect -y 1`

And see if there is a 0x40 in the result.
If not, It might because you are using a external power, instead of the DC-DC Step down converter. In this case, you might need to connect the GND between Raspberry Pi  and Servo Controller, like the orange wire below:
[image]

javierrasp2016, 2016-09-14:
Hi,

Please find below the step I followed. What I need to do? Please let me know if you need me to send more images. Many thanks. Javier.

pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberrypi:~ $ cd Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/
pi@raspberrypi:~/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi $ cd server
pi@raspberrypi:~/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server $ sudo py    thon cali_server.py
Traceback (most recent call last):
  File "cali_server.py", line 5, in <module>
    import motor
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/motor.py"    , line 24, in <module>
    p = pwm.PWM()
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.p    y", line 55, in __init__
    self.set_all_value(0, 0)
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.p    y", line 124, in set_all_value
    self._write_byte_data(self._ALL_LED_ON_L, on & 0xFF)
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.p    y", line 68, in _write_byte_data
    self.bus.write_byte_data(self.address, reg, value)
IOError: [Errno 5] Input/output error

[servo controller_bb]<https://cloud.githubusercontent.com/assets/7285745/18507574/7fa8c282-7aa3-11e6-91bb-5346726490b5.png>

-

javierrasp2016, 2016-09-14:
Hi again,

I was wondering if the problem might be because I run cali_client on my raspberry through putty the first time I did it. I did that through putty opening a new terminal and  running sudo nano cali_client.py and the host shows my Raspberry IP address. Please find the code below. Many thanks again. Javier.

#!/usr/bin/env python
# -*- coding: utf-8 -*-
from Tkinter import *
from socket import *      # Import necessary modules
import os

top = Tk()   # Create a top window
top.title('Raspberry Pi Smart Video Car Calibration')

HOST = '192.168.1.72'    # Server(Raspberry Pi) IP address
PORT = 21567
BUFSIZ = 1024             # buffer size
ADDR = (HOST, PORT)

tcpCliSock = socket(AF_INET, SOCK_STREAM)   # Create a socket
tcpCliSock.connect(ADDR)                    # Connect with the server

runbtn = 'Run'

^G Get Help  ^O WriteOut  ^R Read File ^Y Prev Page ^K Cut Text  ^C Cur Pos
^X Exit      ^J Justify   ^W Where Is  ^V Next Page ^U UnCut Text^T To Spell

sunfounder, 2016-09-18:
Hi @javierrasp2016,
I am afraid that I'm not quite sure where the problem is.
It ran perfectly fine when we test it.
Although, you can try these:
1. Try reboot your raspberry pi, rewire the Servo Controller and try again.

2. Add debug information.
Add a line in PCA9685.py:

    print 'Write value "%2X" to address "%2X" at register "%2X"' % (value, self.address, reg)
before line 68, in _write_byte_data: 

    self.bus.write_byte_data(self.address, reg, value)
like this:
[image]
And run cali_server.py again.
It should print out something this:
[image]
If not, or anything else, please let us know.

javierrasp2016, 2016-09-18:
Hi all again [😊] ,

I have tried the last two instructions and it didn´t work. It gives me the same error.

Having a review of the manual I was thinking about the step in page 46 Basic Software Environment Operation on PC. I didn´t do it because I´m using putty and my laptop. Is what I´ve done right? Should I format the sd car and start from the scratch? I lost honestly. I´ve tried the camera and it worked with the screen and with my laptop.

Just in case please see below the message from cali_client run in my laptop even I know it doen´t get connected with the raspberry pi:

Python 2.7.12 (v2.7.12:d33e0cf91556, Jun 27 2016, 15:19:22) [MSC v.1500 32 bit (Intel)] on win32

Type "copyright", "credits" or "license()" for more information.
>>>
 RESTART: C:\Users\equipo\Downloads\RaspberryPI - Smart CAR\Sunfounder\Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi-master\client\cali_client.py

Traceback (most recent call last):
  File "C:\Users\equipo\Downloads\RaspberryPI - Smart CAR\Sunfounder\Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi-master\client\cali_client.py", line 16, in <module>
    tcpCliSock.connect(ADDR)                    # Connect with the server
  File "C:\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 10061] No se puede establecer una conexión ya que el equipo de destino denegó expresamente dicha conexión
>>>

Many thanks again.

Javier.

  1.  Add debug information.

[qq 20160918133844]<https://cloud.githubusercontent.com/assets/7285745/18613125/3ffe1e8e-7da5-11e6-801f-87700fe0e99d.png>
[qq 20160918134139]<https://cloud.githubusercontent.com/assets/7285745/18613136/a5da4412-7da5-11e6-98ac-bdfc0d87abc6.png>

sunfounder, 2016-09-19:
I am so sorry @javierrasp2016 , If every thing didn't work, it might be the hardware problem. Please email us to get a new Servo Driver. Our email: service@sunfounder.com. I'll give them a heads up.

javierrasp2016, 2016-09-19:
Hi all,

I am sending this email following intructions from Sunfounder.
Please find the email below.
I have been following instruction but it didn't work and it looks the problem might be in the servo driver. So this email is for you to provide a new servo driver because it might be hardware problem.
Please let me know what the next step.

Many thanks.

Javier.

-------- Original message --------
From: SunFounder
Date:19/09/2016 02:18 (GMT+00:00)
To: sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi
Cc: javierrasp2016 ,Mention
Subject: Re: [sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi] invalid syntax (#3)

I am so sorry @javierrasp2016<https://github.com/javierrasp2016> , If every thing didn't work, it might be the hardware problem. Please email us to get a new Servo Driver. Our email: service@sunfounder.com<mailto:service@sunfounder.com>. I'll give them a heads up.

sunfounder, 2016-09-20:
Hi, @javierrasp2016 
Please send us your address for the new servo driver. And please send it with your own email. We can't reply to the email from github.

javierrasp2016, 2016-09-20:
Hi all,

Yes, this is my email account:
jgd82@hotmail.com

My full name is:
Javier Guerrero Dominguez

My address is:
77b Wilkins Road
OX4 2JB
Oxford
United Kingdom

Please let me know if you need anything from as the ticket from the kit. I bought it through Amazon.

Many thanks for your attention.

Best regards

Javier

-------- Original message --------
From: SunFounder
Date:20/09/2016 03:03 (GMT+00:00)
To: sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi
Cc: javierrasp2016 ,Mention
Subject: Re: [sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi] invalid syntax (#3)

Hi, @javierrasp2016<https://github.com/javierrasp2016>
Please send us your address for the new servo driver. And please send it with your own email. We can't reply to the email from github.

closed 2017-02-03 by sunfounder