sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #8: Cervo noise.

elelinde, 2016-11-07:
Hello !
I have a problem running the calibration.
When I start cali_server.py my Cervo is making 'noise' and eventually get warm.
I guess this is not some normal behaving...
I've unplugged and plugged everything about 3 times (or more)...
Where did i fail? Any clue?

sunfounder, 2016-11-08:
Did the servo turns to an extreme angle? If is, that means your servo has installed wrong. 
The operation angle did not match the servos angle. Please refer to the Servo Installation part in the manual comes along the kit.  

[image]

[image]

By the way, good job to unplug the servo when it make noises and get warm, or it might burned dead.

elelinde, 2016-11-08:
Thanks for answering !

I kept the cervo unlinked/unattached from everything.
I did calibration. My cervo can 0-180 from left to right.
Still this noise when starting the server.
Could it be the wiring?
Could it be bad setup?
I'm lost.

Regards.

sunfounder, 2016-11-09:
Hi @elelinde  
I don't think it's bad setup or the wiring.
try this:
1. Unattach all the rocker arm from the servo, but keep the wiring.
2. Run cali_servo.py. Now the steering servo should be at 90, so is the pan servo and the tilt servo.
now, see if the servos make noises.
If not, you can now attach the rocker arm and assemble all of then. Make sure the steering wheels is assembled straight, the pan servo is assembled face front and the tilt servo is assembled face up.
Hope this will help you.

elelinde, 2016-12-03:
Sorry for late answer.
I untached all cervo but the steering one. (to try on just 1 and don't damage others)
I unatached the arm from it.
Stil, when I run cali_server.py the cervo is making that noise.
I really don't get it

sunfounder, 2016-12-05:
Ok, @elelinde I still need to ask a few questions
1. Did all the servo make noise or just one?
2. Did the servo make noise when it is unattached from the arm
3. Is it the noise like high speed motor spinning,  or is it like cracking
4. Can you spin the servo smoothly with gear sound if you unplug the servo wires, or did it crack or easily spins or stuck?
5. when the servo make noises, have you try gentle tap the servo, like for the tilt servo, gentle tap or swing the rocker arm. 
6. Also, may i have a picture of your servos, just check.
Please try all and respond all above in detail. we'll be glad to help you

elelinde, 2016-12-05:
1- All the servo are making noise.
2- Servo makes noise even with arm off.
3- Much more like a cracking sound.
4- When unplugged, the servo can be spinned easily.
5- When the servo is making noise, gently taping on it has no effect. Also, I can't spin them manually.

[image]
[image]
[image]

sunfounder, 2016-12-07:
Hi, @elelinde sorry for being late.
Your servos seems broken. 
Please send us a email with your full name, address and phone number.
We will sent you replacements.
Our email address is: support@sunfounder.com

closed 2017-02-03 by sunfounder