sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #11: Using git submodule for mjpg-streamer

ljdelight, 2016-12-28:
Hey, what do you think about using a git submodule for mjpg-streamer? I ask because the source code for mjpg-streamer in this project is 2+ years old; using a module removes the source from your repo, and makes it very easy to update to the later commits.

A quick overview of the change: the `mjpg-streamer` directory at your root directory would be replaced using a module to https://github.com/jacksonliam/mjpg-streamer. Docs would need changed a bit to say `git clone --recursive` and require the `cmake` package.

sunfounder, 2016-12-28:
Thanks for you head-up 
But sorry, as for so many amateurs using our kit, we prefer a stable version, even though it wouldn't be the best, but not many trouble for amateurs. So if you prefer and able to update the mjpg-streamer, you might want to update one for yourself.
However, we will consider adding this to the Readme. And thanks again

ljdelight, 2016-12-28:
Understandable and it's not too terrible to have the external source living in the project. Sadly the svn repo looks abandoned and was last updated Oct 2014. The jacksonliam/mjpg-streamer is the most active repository that I could find and it has quite a few fixes over the years but the maintainer is using only the experimental directory; their single-dir approach is (in my opinion) the best way to do things since "stable" is defined by a commit and not a directory name.

Thank you for the reply.

closed 2016-12-28 by ljdelight