sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #13: It won't turn

Zombkilr86, 2017-01-02:
When I start up the app on my phone, it connects to the video car just fine. I can move it forwards and reverse, pan the camera every way possible. But when I tilt my phone, the video car does not turn. Do you think it is because I have a raspberry pi 2 model b+ and not a raspberry pi 3 model b+? Or could it be the fact that the app was created for an older version of android? Or is my Samsung Galaxy Amp 2 just not compatible?

sunfounder, 2017-01-05:
Can you copy the output of the Raspberry Pi, I will check it for you.

genath6, 2018-02-15:
i think the problem is on ur phone coz it doesn't have the gravity sensor. I had the same problem with my amazon fire 8d, everything works but i couldnt turn.