sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #14: Error after run the client program

mostafa1981, 2017-01-09:
Hi my name is mostafa!
at the very end of instruction, I run the server program first, the the client program and I can see the graphical user interface of smart car, but when I hit home, left, right or other key, I got the error on the server terminal on raspberry pi that says:"cannot recognize the command:home" or "cannot recognize the command forward"...
I will appreciate if you help me on this

sunfounder, 2017-01-10:
Hi mostafa!
Did you run the right server for the client? I mean, the client_App.py should raun with server.py, and cali_client.py should run with cali_server.py

Also, we prefer question asks on our forum, so, if there is future questions please visit our forum. Thanks

HyLite, 2018-09-23:
sunfounder:
  Your forum is broken. Requires reCAPTCHA response, but reCHAPTCHA not shown on page.

pigmin, 2018-09-24:
Hi. I confirm. Recaptcha v1 api is no longer supported by Google.