sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #15: Cali Server Error

mauriziotallarico, 2017-01-29:
Hello could you please help ion this error ?
thanks !

```
Traceback (most recent call last):
  File "cali_server.py", line 174, in <module>
    setup()
  File "cali_server.py", line 45, in setup
    video_dir.setup()
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/video_dir.py", line 31, in setup
    pwm = servo.PWM()           # Initialize the servo controller.
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 95, in __init__
    self.write_all_value(0, 0)
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 174, in write_all_value
    self._write_byte_data(self._ALL_LED_ON_L, on & 0xFF)
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 113, in _write_byte_data
    if not self._check_i2c():
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 130, in _check_i2c
    status, output = commans.getstatusoutput(cmd)
NameError: global name 'commans' is not defined
```

mauriziotallarico, 2017-01-30:
Hello, porblem solved , i think there is a typo in PCA9685.py : there is a "commans" instead of "commands".
Problem solved,  i think you might want to fix the typo :)

sunfounder, 2017-02-03:
Thank you, We have already fix the problem in the newest commit, you can try pull from master again.

vknabel, 2017-02-04:
Hey we cloned the repo today and ran into trouble when trying to start the `cali_server.py`.

```bash
$ sudo python cali_server.py
offset_x = 11
offset_y = 0
offset = 0
turning0 = True
turning1 = True
Traceback (most recent call last):
  File "cali_server.py", line 174, in <module>
    setup()
  File "cali_server.py", line 45, in setup
    video_dir.setup()
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/video_dir.py", line 31, in setup
    pwm = servo.PWM()           # Initialize the servo controller.
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 113, in __init__
    self.bus = smbus.SMBus(self.bus_number)
IOError: [Errno 2] No such file or directory
```

I read a bit through #3 and accordingly ran:

```bash
$ i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
```

We are using a Raspberry Pi B+ using the latest Raspbian Lite image and we are connecting over SSH.

mauriziotallarico, 2017-02-04:
Hello , i believe i had a similar problem. In my case it was due to the fact that i was powering the Rpi via a different source and not from the batteries of the robot kit. It's mentioned in start page from Git : " **Hardware problem:
If you are using a DC adapter for your Raspberry Pi, you need to connect the GND between Servo Controller and Raspberry Pi. Because they are common-grounded by the micro USB, and if you use a DC adapter, you have to make another wire for common-ground."**

vknabel, 2017-02-04:
@mauriziotallarico Thanks for your fast reply. We power the Pi using the USB of the DC-DC converter which is powered by the kit's batteries. 
We connected the Pi's Ground with the Servo Controller (as it was part of the booklet; GPIO6).

mauriziotallarico, 2017-02-04:
Hello, silly question: i suppose You've enable I2C right ? You can do this by raspi-config

vknabel, 2017-02-04:
@mauriziotallarico Oh god, I guess silly questions are good ones. Somehow we forgot this. 
Thanks a lot!

Sadrpour, 2017-02-17:
I am getting the same error
I checked all the wiring (i followed the instruction on youtube and the notebook that came with robot)
i am using the batteries
i enabled I2C via raspi-config
and i still get the same error 

pi@raspberrypi:~/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server $ lsmod | grep i2c
i2c_bcm2708             4920  0 
i2c_dev                 5671  0 

pi@raspberrypi:~/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server $ sudo python cali_server.py 
offset_x = 11
offset_y = 0
offset = 0
turning0 = True
turning1 = True
Traceback (most recent call last):
  File "cali_server.py", line 176, in <module>
    setup()
  File "cali_server.py", line 47, in setup
    video_dir.setup(busnum=busnum)
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/video_dir.py", line 34, in setup
    pwm = servo.PWM(bus_number=busnum) # Initialize the servo controller.
  File "/home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server/PCA9685.py", line 113, in __init__
    self.bus = smbus.SMBus(self.bus_number)
IOError: [Errno 2] No such file or directory

Sadrpour, 2017-02-17:
issue is fixed ... i had to restart my raspi after enabling the I2C !!!

closed 2017-02-27 by sunfounder

mountainlion0523, 2018-05-21:
@mauriziotallarico you saved my life! thanks

XimikZ, 2018-06-01:
Hi All/ I need Help? After install 2 X 18650 3,7 V, my Pi don't work, Red indicator don't light, green  light, but not long. I cant star terminal PuTTy, bekause low energy for RaspberryPi, Maybe somebody now what problem?

sunfounder, 2018-06-02:
Hi 
Thanks for contacting us,ok,I have asked our engineers help can then reply you asap

Best regards

Jake 

support@sunfounder.com
 
From: XimikZ
Date: 2018-06-01 21:10
To: sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi
CC: SunFounder; State change
Subject: Re: [sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi] Cali Server Error (#15)
Hi All/ I need Help? After install 2 X 18650 3,7 V, my Pi don't work, Red indicator don't light, green light, but not long. I cant star terminal PuTTy, bekause low energy for RaspberryPi, Maybe somebody now what problem?