sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #18: Car not moving straight

raeesmulla, 2017-03-01:
I purchased your car kit, assemble it properly and I am using raspberry pi 3 ,I first calibrated the car but it is not going straight it always take slight curve! Please suggest

sunfounder, 2017-03-02:
Sorry, this is the trouble of the hardwares. The wheels is a little bit loose, also the servo is not a accurate one, so it will never be exactly straight.

raeesmulla, 2017-03-02:
Is there any way to make it straight ,as I need it for my project , one thing I have not enabled i2c will it affect for this straight issue

sunfounder, 2017-03-02:
No, not enabling i2c, the script will not run without errors. i2chelper will help you enable i2c.
And it is not that easy to make it straight. the first thing you will need to do is to fill the gap between the steering  arm and the front wheels, but also keep it smoothly turns, maybe a bearing.
Then, you might need to change the servo to a more accurate one, it might  be expensive.

SebastiaanPasterkamp, 2017-03-05:
I would look into a [PID controller](https://en.wikipedia.org/wiki/PID_controller) for this purpose. Basically it helps to automatically adjust the steering error without the need of constant calibration.