sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #19: motor is unable to move

AAli77, 2017-04-16:
I had build the car and as instructed install and configure all the software and parameter.  I had started server and client which is successfully connected for calibration test when i click on run the car does not move.

Can you please help me that.

thanks

Amir

sunfounder, 2017-04-24:
@AAli77 Sorry for being so late.
Did you power the Pi separately? Did the batteries die out?

AAli77, 2017-04-24:
i had figure out what was the issue but i had a another issue right and left for the car is not working properly and same for the camera right and left is not working properly but top and bottom looks good.

can you helpp on this

AA

sunfounder, 2017-04-25:
@AAli77 please describe the issues, how is it not working properly

AAli77, 2017-04-25:
for let and right it just make turn left and right turn does not work

and foe camera when i press button to left looks good and when i tuen right  ijust rotate 180 and stop.

AA

sunfounder, 2017-05-16:
I am still not quite sure about your question, if you can send us a video to our email, that will be nice.

sunfounder, 2017-05-16:
Oops, our email: service@sunfounder.com

AAli77, 2017-05-29:
I had posted the video's on youtube please have a look.  I had also order 2 more servo motor and also the issue to save the calibration file

Youtube videos:
here is the video for my car
https://youtu.be/EZY6ijXS8oY
https://youtu.be/b2AhsIVv6xw
https://youtu.be/bbtQjcWGgqI﻿

thanks

AA

sunfounder, 2017-06-14:
Hi @AAli77 
I think our support team have replied you, so i am gonna close this issus, is that ok?

AAli77, 2017-09-19:
Sorry for the late reply, I got all the things working.  you can close the issue.

AA

HyLite, 2018-09-05:
Oh, just great! You find  solution but you don't share it with anybody else. What is it about the purpose of community forums you just don't get? Or, is it all take and never give back that clearly defines you?

chiadikav, 2022-05-18:
> I had build the car and as instructed install and configure all the software and parameter. I had started server and client which is successfully connected for calibration test when i click on run the car does not move.
> 
> Can you please help me that.
> 
> thanks
> 
> Amir
How did you fix this?