sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #29: git clone on a pi requires authentication

neilyoung, 2018-01-21:
I'm trying to clone the Sunfounder Kit on a headless PI (connected via SSH from a Mac).

`pi@raspberrypi:~ $ git clone https://github.com/sunfounder/Sunfounder_Smart_Video_Car_ Kit_for_RaspberryPi`

I'm getting asked for a username and password for the repo, which I of course don't have...

What to do?

aliasbox, 2018-01-23:
on a YouTube tutorial I remember hearing psswd is "pi" and/or "raspberry" or "raspbian" (don't remember which video)

Coder2444, 2018-04-09:
which is which. What is the username and what is the password

rfisher001, 2019-12-31:
When you copy and paste from the PDF it puts a space in there. Take the space out and then it won't ask for the password.

closed 2019-12-31 by neilyoung