sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #31: Vibrations and weak construction of the camera mount

neilyoung, 2018-01-21:
I'm having a strange experience with the camera mount. Even if I calibrate it correctly, once the server starts, the camera literally falls down. I need to calibrate it to point to the sky so that it points forward, once the server starts.

But this is not the main issue I have. The entire construction is constantly vibrating and it is absolutely not robust enough. I had a little accident and all the screws fixing the holder of the bottom servo have been pulled out. 

But what is this with the vibrations? If I pan a little left or right, the vibration stops. This is pretty strange, as if the servo would constantly getting contradictory orders

neilyoung, 2018-01-21:
The servo was defective. Replaced. Cheap crap, as it seems

genath6, 2018-02-15:
did the replacement of the servo clear the shaking? bcoz i have the same problem and wanna know how to avoid it coz its the only problem i have ryt now
advance thx

neilyoung, 2018-02-15:
Partly. The massive vibrations were gone, but there are still sometimes situations, when the servos are producing strange howling noises. Meanwhile I have removed the camera mount and glued a RPI fisheye cam, which produce good results. The entire concept of steering by servos, which need to be calibrated, is not working well. I think they would be better off with a fixed gear and a steering which controls left and  right wheels. Will give it a try on occasion.