sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #32: Android app does not work

neilyoung, 2018-01-21:
Not sure, if it is an issue, which needs to be reported here, but the Android app does not work at all. 

The python client app script works fine.

neilyoung, 2018-01-21:
Here is what happens on the Pi. Shortly after that the Android app pops an error: "Network error with ip address ..."

```
Waiting for connection...
...connected from : ('192.168.189.29', 46321)
Command Error! Cannot recognize command: GET /runmode HTTP/1.1
User-Agent: Dalvik/2.1.0 (Linux; U; Android 8.0.0; Nexus 6P Build/OPR5.170623.011)
Host: 192.168.189.28:8000
Connection: Keep-Alive
Accept-Encoding: gzip

```

molexuse, 2018-01-22:
I also do not work

neilyoung, 2018-01-23:
Is here somebody of Sunfounder who cares?