sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #35: BMP280 trows an error on bus types.

BillRothVMware, 2018-08-14:
Leaving a note as requested. I have a Pi Model 3 + 2017 version. I get an the following 280
Traceback (most recent call last):
  File "31_bmp280.py", line 251, in <module>
    main()
  File "31_bmp280.py", line 232, in main
    bmp = BMP280()
  File "31_bmp280.py", line 108, in __init__
    self.bus_number = self._get_bus_number()
  File "31_bmp280.py", line 134, in _get_bus_number
    raise ValueError('Reading Pi type error, Your Pi "{0}"" is not in the list.\n  Please post an Issus at our Github Page or contract us\n    Github page: https://github.com/sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/issues\n    Email: support@sunfounder.com\n    SunFounder'.format(pi_type))
ValueError: Reading Pi type error, Your Pi "Unknown"" is not in the list.
  Please post an Issus at our Github Page or contract us
    Github page: https://github.com/sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/issues

martonz, 2019-01-20:
I'm getting same error message, any solution?

livelyn782295, 2019-02-24:
Traceback (most recent call last):
  File "31_bmp280.py", line 251, in <module>
    main()
  File "31_bmp280.py", line 232, in main
    bmp = BMP280()
  File "31_bmp280.py", line 109, in __init__
    self.bus_number = self._get_bus_number()
  File "31_bmp280.py", line 134, in _get_bus_number
    raise ValueError('Reading Pi type error, Your Pi "{0}"" is not in the list.\n  Please post an Issus at our Github Page or contract us\n    Github page: https://github.com/sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/issues\n    Email: support@sunfounder.com\n    SunFounder'.format(pi_type))
ValueError: Reading Pi type error, Your Pi "Unknown"" is not in the list.
  Please post an Issus at our Github Page or contract us
    Github page: https://github.com/sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/issues
    Email: support@sunfounder.com
    SunFounder

simpsus, 2019-06-15:
I am also a user who gets this error. From looking at the code, the GPIO.RPI_INFO command returns a type "unknown" which cannot be found in the list the script is looking for. There are only two possible modes, 0 and 1 and 0 is for pi model 1, so it is safe for me to replace the _get_bus_number function code with a return 1. Now this runs through.

There is still an error and I am puzzled that my kit contained a BMP180 (not a 280) hardware sensor, but that is beyond this issue ...

sunfounder, 2019-06-17:
Hi Bastian,
Thank you for getting in touch.
Could we know which kit you are using? could you send us your order number and the purchase link? our other staff will help you with it. 
Thanks.
Best Regards,
SunFounder Team
From: Bastian Kennel
Date: 2019-06-16 00:25
To: sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi
CC: Subscribed
Subject: Re: [sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi] BMP280 throws an error on bus types. (#35)
I am also a user who gets this error. From looking at the code, the GPIO.RPI_INFO command returns a type "unknown" which cannot be found in the list the script is looking for. There are only two possible modes, 0 and 1 and 0 is for pi model 1, so it is safe for me to replace the _get_bus_number function code with a return 1. Now this runs through.
There is still an error and I am puzzled that my kit contained a BMP180 (not a 280) hardware sensor, but that is beyond this issue ...