sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #42: [Errno 121] Remote I/O error

Kishwar, 2019-05-26:
Hello Sunfounder,

I am getting following error. Could you please help me on this?

Thanks.

pi@raspberrypi:~/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server $ python cali_server.py 
offset_x = 11
offset_y = 0
offset = 0
turning0 = True
turning1 = True
[Errno 121] Remote I/O error

Your Pi Rivision is: 3 Module B
I2C bus number is: 1
Checking I2C device:
Commands "ls /dev/i2c-1" output:
/dev/i2c-1
I2C device setup OK
Your PCA9685 address is set to 0x40
i2cdetect output:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
Conneceted i2c device:
  0x40
  0x70
Wierd, I2C device is connected. Try to run the program again. If the problem's still, email the error message to service@sunfounder.com
pi@raspberrypi:~/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/server $

Kishwar, 2019-05-26:
Solved. It was power problem.

RomanPanaget, 2022-11-03:
@Kishwar can you explain a little bit more ?