sunfounder/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi #44: struct statx_timestamp

Ann-Rachel, 2019-07-21:
mjpg-streamer 
run: make USE_LIBV4L2=true clean all

In file included from /usr/include/arm-linux-gnueabihf/sys/stat.h:446,
                 from utils.c:33:
/usr/include/arm-linux-gnueabihf/bits/statx.h:25:8: error: redefinition of ‘struct statx_timestamp’
 struct statx_timestamp
        ^~~~~~~~~~~~~~~
In file included from utils.c:32:
/usr/include/linux/stat.h:56:8: note: originally defined here
 struct statx_timestamp {
        ^~~~~~~~~~~~~~~
In file included from /usr/include/arm-linux-gnueabihf/sys/stat.h:446,
                 from utils.c:33:
/usr/include/arm-linux-gnueabihf/bits/statx.h:36:8: error: redefinition of ‘struct statx’
 struct statx
        ^~~~~
In file included from utils.c:32:
/usr/include/linux/stat.h:99:8: note: originally defined here
 struct statx {
        ^~~~~
make: *** [<builtin>: utils.o] Error 1

I try to downgrade libc6 but it is not possible and stuck on version 2.28.

Any solution? :/

arminf82, 2019-07-21:
How did you install it? Did you follow the instructions?

https://wiki.ubuntuusers.de/MJPG-Streamer/

Ann-Rachel, 2019-07-21:
I follow the description on their page:
https://www.sunfounder.com/learn/Smart-Video-Car-for-Raspberry-Pi/56-60-smart-video-car.html

sudo apt-get install subversion
sudo apt-get install libv4l-dev
sudo apt-get install libjpeg8-dev
sudo apt-get install imagemagick

Compile the source code of MJPG-streamer:
cd /home/pi/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/mjpg-streamer/mjpg-streamer
sudo make USE_LIBV4L2=true clean all

follow the link, I get the same error...

arminf82, 2019-07-21:
Try to just 

Make

Without USE_LIBV4L2=true.

Should work if you don't have a fancy webcam and use the provided one.

I will try to install mjpg-streamer on a different machine again keep you updated.

Ann-Rachel, 2019-07-24:
thank you, but the camera still doesn't work. I use the one from the set... nothing special...

fgcjamin, 2019-08-12:
Hello,
I had the same issue, I commented the concerned includes in mjpg-streamer/utils.c line 32 & 33 :
//#include <linux/stat.h>
//#include <sys/stat.h>
Then
>>make clean all
and it works.

zhangzhiyong0410, 2019-08-13:
This is caused by incompatibility with the new Raspberry Pi system.

marvimation, 2020-04-17:
> Hello,
> I had the same issue, I commented the concerned includes in mjpg-streamer/utils.c line 32 & 33 :
> //#include <linux/stat.h>
> //#include <sys/stat.h>
> Then
> 
> > > make clean all
> > > and it works.

Sounds great  fgcjamin! How did you comment out the concerned includes? Do you cd into the mjpg-streamer utils somehow? Please explain your procedure for the noobies thanks.

Willyzw, 2020-05-04:
> > Hello,
> > I had the same issue, I commented the concerned includes in mjpg-streamer/utils.c line 32 & 33 :
> > //#include <linux/stat.h>
> > //#include <sys/stat.h>
> > Then
> > > > make clean all
> > > > and it works.
> 
> Sounds great fgcjamin! How did you comment out the concerned includes? Do yo

In terminal, you can navigate to the desired folder by `cd ~/Sunfounder_Smart_Video_Car_Kit_for_RaspberryPi/mjpg-streamer/mjpg-streamer` and then edit it by `nano utils.c`

marvimation, 2020-05-15:
Cool! I will try it! Thanks for your kind help!

Best always!

Marco

srikanthaddala, 2020-05-22:
> Hello,
> I had the same issue, I commented the concerned includes in mjpg-streamer/utils.c line 32 & 33 :
> //#include <linux/stat.h>
> //#include <sys/stat.h>
> Then
> 
> > > make clean all
> > > and it works.

this worked for me too thanks